NilujePerchut/kicad_scripts #2: Script not working with kicad rev  2018-02-10 revision a04965c36

B-Kohl, 2018-02-11:
I try to use your teardrop script. Using the plugin (Tools-External Plugins), the "OK" button seems to do nothing.

So I started the script using the Python Shell:

```
td.SetTeardrops(30,70,10,None);
+Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/td.py", line 227, in SetTeardrops
    pcb.Add(__Zone(pcb, coor, track))
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/td.py", line 95, in __Zone
    z.BuildFilledSolidAreasPolygons(board)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 12661, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, ZONE_CONTAINER, name)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'ZONE_CONTAINER' object has no attribute 'BuildFilledSolidAreasPolygons'

td.SetTeardrops()
+Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/td.py", line 227, in SetTeardrops
    pcb.Add(__Zone(pcb, coor, track))
  File "C:\Program Files\KiCad\share\kicad\scripting\plugins/td.py", line 95, in __Zone
    z.BuildFilledSolidAreasPolygons(board)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 12661, in <lambda>
    __getattr__ = lambda self, name: _swig_getattr(self, ZONE_CONTAINER, name)
  File "C:\Program Files\KiCad\lib\python2.7\site-packages/pcbnew.py", line 80, in _swig_getattr
    raise AttributeError("'%s' object has no attribute '%s'" % (class_type.__name__, name))
AttributeError: 'ZONE_CONTAINER' object has no attribute 'BuildFilledSolidAreasPolygons'
```

Kicad version:
Application: kicad
Version: (2018-02-10 revision a04965c36)-makepkg, release build
Libraries:
    wxWidgets 3.0.3
    libcurl/7.54.1 OpenSSL/1.0.2l zlib/1.2.11 libssh2/1.8.0 nghttp2/1.23.1 librtmp/2.3
Platform: Windows 7 (build 7601, Service Pack 1), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.3 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.60.0
    Curl: 7.54.1
    Compiler: GCC 7.1.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_SPICE=ON

NilujePerchut, 2018-02-12:
Thank you for this report.
It seems that the python API changed again. I will investigate to get the new way to build filled polygons.

NilujePerchut, 2018-02-18:
After a quick research, I found out that the zone filling code has been deeply reworked. In order to minimize the work load I will wait until the V5 release to find a full fix (this specific part is always moving).
For now, I have disabled the teardrop zone filling, you just need to press b to fill the teardrops after the script execution.

I will keep this issue open as a reminder. When the V5 is out, I will find a fix and close it for good.

Sorry for this problem.

obra, 2018-07-27:
it looks like v5 is out. Once you have a fix, I'm very much looking forward to trying it. Thanks so much for working on it!

NilujePerchut, 2018-07-31:
I'm on it. You may expect some results next week.
From what I have read on the Kicad.info forum, I need to switch from wxpython to tkinter. It is not a big deal but their might be some surprises on the way.

NilujePerchut, 2018-08-09:
Here are updates on it:

The code to fill zones is not back in Kicad V5. So there is actually no way to fill zones using the Python API.
Fortunately, it will come back with V5.0.1 (which should also correct the wxPython problem). See https://lists.launchpad.net/kicad-developers/msg37083.html for more details.

If you really need to use this scripts now, just use it normally and press the "b" key after each teardrop add.

obra, 2018-08-09:
Awesome. Thanks so much for the update! (And indeed, having to rebuild
my zones after just about any change is a pretty reasonable thing :)

W-M-D, 2019-03-15:
any hope for a fix on this ?

NilujePerchut, 2019-03-16:
I am sorry about the long silence.
I was waiting the 5.1.0 release before doing anything.
Since it is now out, you can can expect news during the next week.

W-M-D, 2019-03-18:
Awesome!

NilujePerchut, 2019-03-24:
Done. Hope it will help.

closed 2019-03-24 by NilujePerchut