NilujePerchut/kicad_scripts #3: Segfault on Kicad 5 (nightly build)

leoheck, 2018-09-27:
Hi! I am trying to use your script with Kicad 5. 

I installed it as you said but I just received a segfault starting the PCBNew. Running it with gdb I got this output:

```
Thread 1 "kicad" received signal SIGSEGV, Segmentation fault.
0x00007fffcd20fbb7 in wxStatusBarBase::CalculateAbsWidths(int) const () from /usr/lib/x86_64-linux-gnu/libwx_gtk3u_core-3.0.so.0
```

Do you know if it is something related with the teardrop script or just a random issue related to the plugin system? I am asking to deside if I can report it on launchpad for Kicad developers.

This is the version I am using if you would like to know.

```
Application: kicad
Version: 5.0.1-dev-unknown-1f19903~66~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.0g zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 4.15.0-34-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 2.24
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.3.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_WXPYTHON=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON
```

NilujePerchut, 2019-03-24:
Version 0.4.2 corrects it.
Sorry for the delay.

closed 2019-03-24 by NilujePerchut

leoheck, 2019-03-25:
Cool, thanks