NilujePerchut/kicad_scripts #4: Error

epubc, 2019-04-19:
When I run the Teardrop Plugin on Kicad v5.1 (Windows 10) an error occurred:

> Exception on python action plugin code
---------------------------
Traceback (most recent call last):

  File "C:\Users\ePub\AppData\Roaming\kicad\scripting\plugins/teardrops/teardrop_plugin.py", line 22, in Run
    InitTeardropDialog(GetBoard())

  File "C:\Users\ePub\AppData\Roaming\kicad\scripting\plugins/teardrops/teardrop_dialog.py", line 55, in InitTeardropDialog
    tg = TeardropDialog(board)

  File "C:\Users\ePub\AppData\Roaming\kicad\scripting\plugins/teardrops/teardrop_dialog.py", line 18, in __init__
    super(TeardropDialog, self).__init__(None)

  File "C:\Users\ePub\AppData\Roaming\kicad\scripting\plugins/teardrops/teardrop_gui.py", line 22, in __init__
    self.SetSizeHints( wx.DefaultSize, wx.DefaultSize )

  File "C:\Program Files\KiCad\lib\python2.7\site-packages\wx-3.0-msw/wx/_core.py", line 9718, in SetSizeHints
    return _core_.Window_SetSizeHints(*args, **kwargs)

TypeError: in method 'Window_SetSizeHints', expected argument 2 of type 'int'

---------------------------
OK   
---------------------------

NilujePerchut, 2019-04-21:
I am afraid the windows version of Kicad is using Python2. Which is not supported by this script anymore.
I will check if it is possible support both Python2 and Python3 versions but I cannot promise anything.

bercebus, 2019-05-06:
Hi,

So, do you know any version of Windows with python 3 activated to be able to use your plugin?

NilujePerchut, 2019-05-06:
Hi,
Unfortunately, I know no version of Kicad for Windows with python3 by default.
Ubuntu 18.04 allows you to run kicad with python3 by default. This is fact the problem behind all of this (due to gtk support by python).

NilujePerchut, 2019-05-12:
This must be corrected now (v2.4.4).
Can you test it again ?

epubc, 2019-05-13:
Hi,
thank you very much ,, it worked well

closed 2019-05-13 by epubc

bercebus, 2019-05-13:
What a pity, I thought that the new version would work in Linux with Python 2 as well as in Windows.

JianzhouSu, 2019-05-28:
One possbile solution:
Open the teardrop_gui.py, and replace the "SetSizeHints" with "SetSizeHintsSz".
This error might caused by different version of wxpython used in KiCad Python.

bercebus, 2019-05-28:
> One possbile solution:
> Open the teardrop_gui.py, and replace the "SetSizeHints" with "SetSizeHintsSz".
> This error might caused by different version of wxpython used in KiCad Python.

Thank you very much, JianzhouSu, it was the solution!