NilujePerchut/kicad_scripts #5: Kicad exception

leoheck, 2019-05-07:
Hi @NilujePerchut I am receiving the following exception when opening the PcbNew on Kicad.

I filed a bug for Kicad devs. It is possible that teardrop plugin is causing this, because I have only one plugin.

[image]

I am using Kicad `Version: 5.1.2-f72e74a~84~ubuntu19.04.1, release build`

https://bugs.launchpad.net/kicad/+bug/1826493

NilujePerchut, 2019-05-12:
Hi @leoheck,
I do not think this is coming from my plugin. Nevertheless, I do not have a lot of experience with the python3 version, this could be something I missed.
I noticed you are using Ubuntu 19.04.1, it might be related to it. Did you try on another version of Ubuntu ?
Are you getting the same problem without any plugin at all ?
I just released a new version (2.4.4), even if this should not have any influence on it, can you try it ?

leoheck, 2019-05-13:
Hello, @NilujePerchut I don't have this issue very often but I don't know it is related to your plugin. I will install the new version to monitor if this happens again. Thank you for the feedback.

NilujePerchut, 2019-11-23:
Hello @leoheck , I would like to close this issue.
Do you have any news about it ?

leoheck, 2019-11-24:
Hello, @NilujePerchut I think I am good. I will do it for you, right?

closed 2019-11-24 by leoheck