NilujePerchut/kicad_scripts #6: Directory structure not clone friendly

procsynth, 2019-05-12:
Hi !
Is it possible to put the content of the teardrop directory at the root of the repository ? It makes it easier to just clone it into the KiCad plugin folder (or add it as a submodule). If you do that may want to rename your repo.
Having the files at the root enables easier update by a pull and easy installation.

I could make a pull request if you want.

Thanks for your work,
Antoine

NilujePerchut, 2019-05-12:
Hi,
This repo was dedicated to be home the home of several plugins. The inner folder was designed to avoid mess on the root dir.
Is it a problem to clone the whole repo and then just symlink the teardrop directory in the .kicad_plugins directory ? This method allows selection of desired plugins (when there will be more than one).

I must confess, I don't know if I will write new plugins. Maintaining just one is already not easy.

bercebus, 2019-05-13:
Could I request a new feature in your plugin?

Is it possible to add to make rounded corners on PCB tracks in any way?

procsynth, 2019-05-13:
Hi @NilujePerchut,
I understand, I wanted to do the same thing with my plugin directory but I ended up making a repo per plugin and then a repo collecting all repos of interest as submodule (see https://github.com/stimulu/stimulu-kicad-plugins).
As you can see I integrated a fork of yours (https://github.com/stimulu/kicad-teardrops) as a submodule.
I find it easier to work with in a git environnement but symlink would work in most cases I guess.

procsynth, 2019-05-13:
Hi @bercebus,
Maybe you could be interested by this plugin : https://github.com/stimulu/kicad-round-tracks

bercebus, 2019-05-13:
> Hi @bercebus,
> Maybe you could be interested by this plugin : https://github.com/stimulu/kicad-round-tracks

Thank you a lot, it works very well

NilujePerchut, 2020-07-21:
Fixed in commit 556c4995fa626dc943d09b618379d4821641e381

closed 2020-07-21 by NilujePerchut