NilujePerchut/kicad_scripts #7: Teardrops on vias work but not on pads

sirus20x6, 2019-07-08:
This is probably because I exported a diptrace pcb as an eagle board format so I could import it into kicad. I know that's a weird edge case, but I could try to provide a file that exhibits this behavior with only one part and one trace on it if that's helpful.

leevawns, 2019-08-01:
it can work with pad standar

NilujePerchut, 2019-11-23:
I am sorry to handle this issue only now. I was on best effort mode for the previous months.
From what I see, it is possible to add teardrops on via and standard TH pads.

If you seek teardrops for SMD pads to, check PR #13 or just wait for me to include it in master.

NilujePerchut, 2019-12-16:
PR #13 is now merged to master branch so I am closing this issue.

closed 2019-12-16 by NilujePerchut