NilujePerchut/kicad_scripts #12: Teardrops fail when three traces go to via

aaronw2, 2019-11-14:
When I have three traces going to a via but the teardrops only not created for all of the traces. In the below image, only one of the front traces (going to the right) has a teardrop.
[image]
Here is another via with only one teardrop.
[image]
Note that the teardrops are visible in the gerber view.

Kicad version 5.1.4
wxWidgets 3.0.3

NilujePerchut, 2019-11-15:
If teardrops are visible in the gerber files, I suspect that the missing one is created but not displayed.
This is a known bug which should have been corrected by  #9 .
I will check I can figure out why it is not working for all layers.
Meanwhile, you should be able to make them visible by saving the pcb project and reopen it.

aaronw2, 2019-11-16:
Saving and quitting and reloading didn't work. I suspect it is a Kicad bug
since sometimes I see them all then they disapear.

-Aaron

NilujePerchut, 2019-11-23:
I think you are right, it seems like it is a Kicad bug.
I will close this issue, but feel free to re-open it if you have news about it.

closed 2019-11-23 by NilujePerchut