NilujePerchut/kicad_scripts #15: Teardrop zones not being filled

pdp7, 2019-12-11:
I am using the Teardrop Action plugin.  Thanks for creating it! :)

I noticed that it creates the outline of the teardrop but does not fill the zone.  I feel like this is related to some DRC functionality preventing it from filling to close to the via?

Any ideas would be appreciated!

Thanks,
Drew
[flex-bizcard-teardrop.kicad_pcb|attachment](upload://28ukid1SR3UksogJSmUTVOmOWye.kicad_pcb) (37.5 KB) [image] [image] [image] [image][flex-led-tester-0603-teardrop.kicad_pcb|attachment](upload://kAK3YH5ssCzDng64VdzeOOk7odd.kicad_pcb) (51.4 KB)

NilujePerchut, 2019-12-11:
You're right. The zones are not filled due to DRC rules.
In fact, the root of the problem is that your tracks and pads are not related to any net.
I suspect that you are not using schematic beforehand.

Unfortunately, this teardrop plugin cannot work on pads/tracks without nets. Indeed it would be very dangerous to consider that every pad/track without name has the same net.
The best way for you to solve the problem is to make a schematic or at least a quick netlist and import it in pcbnew.

Cédric

pdp7, 2019-12-16:
Thank you, that seems to have been the issue. I read did with a schematic and then imported the netlist and redid the layout. The teardrop zones now finally fill in :slight_smile:

Screenshot:
https://kicad-info.s3.dualstack.us-west-2.amazonaws.com/original/3X/9/a/9a606da0e2b7a8e3cdcde314b5a0456f5dc6b63b.png

NilujePerchut, 2019-12-16:
Looks nice.
I'm glad to see it is working now.

pdp7, 2019-12-16:
@NilujePerchut I noticed that a one side of vias are not getting teardrops.  Any ideas?

Here is Google Drive with the KiCad schematic, layout and screenshots:
https://drive.google.com/open?id=1m63qP6nTkan94mBolEdnslTogIJEDCDW

Here is a screenshot that shows what I mean:
https://drive.google.com/file/d/1QPvWue11-4IttkACCQw2gi1FoHoIX4YF/view?usp=sharing

The F.Cu trace has a teardrop going into the via but the B.Cu trace going into the via does not a filled teardrop.  It seems something is preventing the zone from being filled.

Thank You.

NilujePerchut, 2019-12-16:
@pdp7 The B.Cu teardrops are not filling because of the text box on the same side.
From what I understand, Kicad is considering the global rectangular box around text instead of the text itself. From my point of view, this is a Kicad bug.
One easy (but not clean) way to overcome such problem is to split the text to not include the teardrop in the text rectangle.
I hope this will help you.

pdp7, 2019-12-17:
@NilujePerchut thanks for pointing this out.  I deleted the text to test and the teardrop zone does now fill in.  For the actual solution, I just had to shift the text down slightly so that it was far enough away from the via.  The teardrops then all filled correctly.

closed 2019-12-17 by pdp7