NilujePerchut/kicad_scripts #16: Teardrop misaligned with SMD pad

pdp7, 2019-12-17:
This is a minor issue but I noticed that the teardrop does not seem to be aligned as good as possible with SMD pad in my flexible business card design.

Here is what the plugin creates:
https://drive.google.com/file/d/1imy8g4_UDSdoS6f8n0J__WEpn0eMUByY/view?usp=sharing

I tweaked the the teardrop zone by moving it to flow better into the SMD pad:
https://drive.google.com/file/d/1lRFflo2ZKCVdB1BJzMa1ozuS1OzSonR4/view?usp=sharing

Here are the default plugin settings that I am using:
https://drive.google.com/file/d/1nGocxgLwbLY6CU8cGp8Qq-fnyXvXXEJm/view?usp=sharing

Here are the Kicad files:
https://drive.google.com/open?id=1imy8g4_UDSdoS6f8n0J__WEpn0eMUByY

pdp7, 2019-12-18:
My collegue @tekdemo made the observation that:
> I'm betting that it doesn't support square pads, so it's using the pad size and treating it as you would a circle.

Here is an image:
https://drive.google.com/open?id=1gMsfAvsQHncMkAzHFHOjQy3PrJKn0-DX

NilujePerchut, 2019-12-18:
You're right.
SMD pad support is just basically treating SMD pads as if they were rounded TH pads/vias.
I will take a lot to see if it is possible to take in account the pad shape. But it seems difficult since Kicad is now allowing custom pad shapes.

pdp7, 2019-12-26:
@NilujePerchut I am not sure how else to send this to you but I wanted to say that I wrote a blog post about using the plugin: https://blog.oshpark.com/2019/12/25/teardrops-in-kicad/

thanks so much for your efforts!  if you ever want flex PCBs for testing, then just email me at drew<AT>oshpark.com

NilujePerchut, 2020-04-10:
I am sorry for this late answer.
And thanks for the blog post !! It is nice to see people using my code.

Concerning the issue, unfortunately, I do not see how to solve it (at least for now). But ideas are welcome.

I will let the issue open in order to remind me that this has to be solved.

PS: I didn't know that oshpark is now doing flex pcb too. I will definitively email you in case I need flex pcb.

fabiangeissler, 2020-05-08:
Hey, thanks for the great work so far!
Can you maybe for now just fetch the most common case of a straight line going vertically or horizontally into a (rounded) square pad? That would be great!

NilujePerchut, 2022-04-16:
I suggest you to try the native Kicad teardrop implementation (V6.99). It already supports what you need.