NilujePerchut/kicad_scripts #18: Teardrops in zones (ground planes)

tmanaud, 2020-04-07:
There is an issue with via with tracks on planes with copper pour.  

Sample with copper pour and vias without teardrops: expected copper pour fill:
[image]

Sample with copper pour and vias with teardrops applied: prevents copper pour
[image]

Note: ground stitching has to be attached with tracks, not just vias otherwise it just create via "island" on the next zone refill.  
Suggested solution: do not place teardrops if via are inside area/copper pour.  

Application: Pcbnew
Version: 5.1.5-52549c5~86~ubuntu19.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.0 OpenSSL/1.1.1b zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh/0.8.6/openssl/zlib nghttp2/1.36.0 librtmp/2.3
Platform: Linux 5.0.0-38-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
    Boost: 1.67.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.64.0
    Compiler: GCC 8.3.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

NilujePerchut, 2020-04-10:
Hi,
I tried to reproduce your problem without success.
Is it possible for you to send me the .kicad_pcb file for me to check ?
Thanks in advance.

tmanaud, 2020-04-11:
Hello NilujePerchut, thanks for taking time to look into this issue.  

Sure, here is the .kicad_pcb file zipped:
[pcb_test.kicad_pcb.zip](https://github.com/NilujePerchut/kicad_scripts/files/4464289/pcb_test.kicad_pcb.zip)

When reproducing the issue, I found out two parts:
- teardrops are created inside ground/area
- when area are filled, with teardrop/via close to area side, it fails.
[image]

Feel free to ask if I can help you in any way.

NilujePerchut, 2020-04-12:
This should be fixed now (v0.4.8). 
I added and option (true by default) to keep the possibility to restore the original behaviour.
This might be useful in the following cases:
1) This new option might slow the process down
2) The behaviour is not defined if multiple zone overlap

Please, test it and let me know.
If it is ok for you, I will close the issue.

tmanaud, 2020-04-13:
Wow, that was an ultra quick fix!
Fixed it on my side:
[image]

Thanks a lot!

closed 2020-04-13 by tmanaud