NilujePerchut/kicad_scripts #20: Teardrop issue in inner layers : Extra teardrops being added in inner layers

chirag070, 2020-05-18:
Hello There,
Thanks for making this wonderful plugin available for everyone. It helps a lot. I have observed one issue as described below.

When tracks are routed in inner layer which are beneath the SMD pads of Top/Bottom layers, The plugin generates teardrops in inner layer. Here are the screenshots and KiCAD project file attached.

[image]

[image]

[teardrop_kicad_files.zip](https://github.com/NilujePerchut/kicad_scripts/files/4642404/teardrop_kicad_files.zip)

KiCAD Version : Stable 5.1.5_3, Windows, 64bit
Plugin Version : 0.4.8

[image]

NilujePerchut, 2020-05-18:
Hi,
Thanks for such a detailed report.
I will fix that as soon as possible.

NilujePerchut, 2020-06-05:
Hi,
Sorry for the delay.
Can you confirm that commit 0bcbbc377f2a5a13c646d1fc5ed16993f87a8f73 fixes you problem ?
Thanks in advance.

chirag070, 2020-06-06:
Hello,
Yes. v0.4.9 solved the issue. Thanks a lot.
[image]

closed 2020-06-06 by chirag070