NilujePerchut/kicad_scripts #23: Include SMD and single via mode don't do anything

Pun-e, 2020-07-08:
**Plugin version:** V0.4.9 installed from github
**OS:** W10 64bit
**KiCad Version** Pcbnew
Version: (5.1.6)-1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.66.0 OpenSSL/1.1.1d (Schannel) zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.1.1) nghttp2/1.39.2
Platform: Windows 8 (build 9200), 64-bit edition, 64 bit, Little endian, wxMSW
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8)
    Boost: 1.71.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.66.0
    Compiler: GCC 9.2.0 with C++ ABI 1013

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=OFF
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

**Installed:** ~\AppData\Roaming\kicad\scripting\plugins\teardrops

**Behaviour 1:**
Select a via or smd pad,
Open plugin using UI button,
Will add teardrops to everything, ignoring initial selection

**Behaviour 2:**
Open plugin using ui shortcut, nothing pre-selected
Check "include smd pads"
Select "ok"
Nothing happens

All the other options seem to work as expected. Unsure If I've installed wrong or an issue with my vesion od KiCad. I've seen the other issue for KiCad V5.99 but I'm pretty sure I am running the stable version.

gcormier, 2020-07-10:
Same issue on KiCad stable on Windows.

NilujePerchut, 2020-07-24:
Thanks for the reports
The 948ad532adca14d073f03e0946d5fa9cd9ab2010 commit should fix this.
Can you confirm ?

gcormier, 2020-07-24:
Still having the issue on a few designs. https://github.com/gcormier/megadesk is one of them experiencing the issue.

NilujePerchut, 2020-07-24:
Thanks for the design. It helps a lot !
Commit 32b85a423e0044b4d814421681419c0dd360b6cb should fix the bug. Can you confirm it is ok for you ?

Nevertheless, teardrops on SMD pads are not fully supported for the moment. Expect strange behavior according to the pad shape/size/orientation.

gcormier, 2020-07-27:
Unfortunately on windows, still no effect clicking the button. Nothing happens.

NilujePerchut, 2020-08-10:
I am sorry to come back to you so late.
I didn't manage to reproduce your problem in a Windows 10 virtual machine.
Are you sure you pulled the latest commits before testing ?
If yes, can you tell me more about your configuration.

gcormier, 2020-08-25:
Sorry, just on vacation but I will try to get more information to you soon!.