NilujePerchut/kicad_scripts #25: KiCad Windows Locked after Closing Teardrops Window (macOS)

zivkovic-msz, 2020-08-02:
When I close the Teardrops window or when it automatically closes after adding the teardrops, all of my KiCad windows become locked and I need to force quit KiCad to close it. Has anyone else encountered this kind of issue?

I am using the latest stable version of Pcbnew with macOS Mojave Version 10.14.6:

Application: Pcbnew
Version: (5.1.6-0-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.54.0 LibreSSL/2.6.5 zlib/1.2.11 nghttp2/1.24.1
Platform: macOS Mojave Version 10.14.6 (Build 18G6020), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.72.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.54.0
    Compiler: Clang 9.0.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

Thanks.

NilujePerchut, 2020-08-10:
I never encountered such behaviour.
But I never tested my script on macOS.
Is it possible for you to send me the design that produce such bug. I will try to reproduce it.

zivkovic-msz, 2020-08-12:
Thanks @NilujePerchut. You can recreate the issue using this [project from HackadayU](https://github.com/wyolum/HackadayU_KiCad_FreeCAD).

AndrewKeene, 2020-09-07:
Hi everyone.  I'm noticing the same behavior on MacOS Catalina with the same KiCad stable release version (5.16-0-10_14).  KiCad requires a force close after adding or removing teardrops.  I tried basic edits to the td.py script without any luck.  This issue may be agnostic to the design as it happens with all my local KiCad projects.

Application: KiCad
Version: (5.1.6-0-10_14), release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.64.1 SecureTransport (LibreSSL/2.8.3) zlib/1.2.11 nghttp2/1.39.2
Platform: macOS Catalina Version 10.15.6 (Build 19G2021), 64 bit, Little endian, wxMac
Build Info:
    wxWidgets: 3.0.4 (wchar_t,STL containers,compatible with 2.8)
    Boost: 1.72.0
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.54.0
    Compiler: Clang 9.0.0 with C++ ABI 1002

Build settings:
    USE_WX_GRAPHICS_CONTEXT=ON
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=OFF
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

NilujePerchut, 2020-09-30:
Hi all,
I finally managed to install a MacOS virtual machine.
I have corrected a bug that prevents the teardrop dialog from closing. I think this can explain the behaviour you are facing.
Can you tell me if commit 4f38e363bc9a95e1769e17bc96d2820c03fd6289 fixes the problem ?

AndrewKeene, 2020-10-02:
Confirmed fixed.  Tested with the new KiCad 5.1.7.

closed 2020-10-02 by NilujePerchut