NilujePerchut/kicad_scripts #26: v5.99:

pdp7, 2020-08-12:
@NilujePerchut not sure if you want issues to be opened for v5.99 since I believe you are just focusing on v5.1 for now.  
[image]
But I thought I would log this just in case.

I am getting a WxPython error for Teardrops v.0.4.9 when I click "OK".  I can't seem to copy the text of the full error so I have made a screenshot.

KiCad version:
```
Application: Pcbnew

Version: (5.99.0-2643-g1bb6d4e22), debug build

Libraries:
	wxWidgets 3.0.4
	libcurl/7.68.0 GnuTLS/3.6.13 zlib/1.2.11 brotli/1.0.7 libidn2/2.2.0 libpsl/0.21.0 (+libidn2/2.2.0) libssh/0.9.3/openssl/zlib nghttp2/1.40.0 librtmp/2.3

Platform: Linux 5.8.0 x86_64, 64 bit, Little endian, wxGTK

Build Info:
	Date: Aug 10 2020 00:05:57
	wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.24
	Boost: 1.71.0
	OCE: 6.9.1
	Curl: 7.68.0
	ngspice: 31
	Compiler: GCC 9.3.0 with C++ ABI 1013

Build settings:
	KICAD_SCRIPTING=ON
	KICAD_SCRIPTING_MODULES=ON
	KICAD_SCRIPTING_PYTHON3=OFF
	KICAD_SCRIPTING_WXPYTHON=ON
	KICAD_SCRIPTING_WXPYTHON_PHOENIX=OFF
	KICAD_SCRIPTING_ACTION_MENU=ON
	BUILD_GITHUB_PLUGIN=ON
	KICAD_USE_OCE=ON
	KICAD_SPICE=ON
	KICAD_STDLIB_DEBUG=OFF
	KICAD_STDLIB_LIGHT_DEBUG=OFF
	KICAD_SANITIZE=OFF
```

NilujePerchut, 2020-08-13:
Thanks for this detailed report.
It seems that devs have broken the zone API once again. I bet this might also explain #22.
You're right, I have not planned to support 5.99 cycle. But I am very interested in the new rule-based DRC. So I will take a look at the problem and see if I can find a workaround that works on both branches.

pdp7, 2020-08-18:
I've looked into this a bit more and I found the commit where the function went away:
```diff
commit 1a7cef2950a89959e76dc7b5de259b3b022a7f2e
Author: Jeff Young <jeff@rokeby.ie>
Date:   Sat Jul 13 17:35:57 2019 +0100

    Improve performance, commenting and API of some polygon classes.

diff --git a/pcbnew/class_zone.h b/pcbnew/class_zone.h
index e42af2637454..75a48d8dc929 100644
--- a/pcbnew/class_zone.h
+++ b/pcbnew/class_zone.h
@@ -247,18 +247,6 @@ public:
      */
     bool HitTest( const wxPoint& aPosition, int aAccuracy = 0 ) const override;

-    /**
-     * Function HitTest
-     * tests if a point is inside the zone area, i.e. inside the main outline
-     * and outside holes.
-     * @param aPosition : the wxPoint to test
-     * @return bool - true if a hit, else false
-     */
-    bool HitTestInsideZone( const wxPoint& aPosition ) const
-    {
-        return m_Poly->Contains( VECTOR2I( aPosition ), 0 );
-    }
-
```

This causes td.py to fail:
```py
def __DoesTeardropBelongTo(teardrop, track, via):
    """Return True if the teardrop covers given track AND via"""
    # First test if the via belongs to the teardro
    if not teardrop.HitTestInsideZone(via[0]):
        return False
    # In a second time, test if the track belongs to the teardrop
    if not track.HitTest(teardrop.GetBoundingBox().GetCenter()):
        return False
    return True
```

I'm trying to see what the new to do it is.

pdp7, 2020-08-18:
I opened an issue to get guidance on what the alternative is to the deleted method:
https://gitlab.com/kicad/code/kicad/-/issues/5241

pdp7, 2020-08-20:
[From developer Jeff Young:](https://gitlab.com/kicad/code/kicad/-/issues/5241)
> `HitTestFilledArea`. You'll need to specify a layer and an accuracy (use 0 to mimic the behaviour of the old routine).
>
> Note that they are slightly different as the old routine would subtract out the edge of the zone -- but I don't think you want that anyway.

pdp7, 2020-08-20:
I believe I was able to get `HitTestFilledArea()` working:
```diff
diff --git a/teardrops/td.py b/teardrops/td.py
index 997e8d6cfd14..0992dc570b96 100755
--- a/teardrops/td.py
+++ b/teardrops/td.py
@@ -76,11 +76,18 @@ def __GetAllTeardrops(board):
 def __DoesTeardropBelongTo(teardrop, track, via):
     """Return True if the teardrop covers given track AND via"""
     # First test if the via belongs to the teardrop
-  `  if not teardrop.HitTestInsideZone(via[0]):
-        return False
+    #if not teardrop.HitTestInsideZone(via[0]):
+    print( track.GetLayer() )
+    print( via[0] )
+    print( teardrop.HitTestFilledArea(track.GetLayer(), via[0]) )
+    #if not teardrop.HitTestFilledArea(track.GetLayer(), via[0])
+        #print("DEBUG if not teardrop.HitTestInsideZone(via[0])")
+    #    return False
     # In a second time, test if the track belongs to the teardrop
     if not track.HitTest(teardrop.GetBoundingBox().GetCenter()):
+        print("DEBUG if not track.HitTest(teardrop.GetBoundingBox().GetCenter())")
         return False
+    print("DEBUG return True")
     return True
```

pdp7, 2020-08-20:
The next issue is with ZONE_FILLER.
```diff
diff --git a/pcbnew/zone_filler.h b/pcbnew/zone_filler.h
index 2427c3892482..d26bd6a6ecad 100644
--- a/pcbnew/zone_filler.h
+++ b/pcbnew/zone_filler.h
@@ -39,7 +39,7 @@ class SHAPE_LINE_CHAIN;
 class ZONE_FILLER
 {
 public:
-    ZONE_FILLER( BOARD* aBoard, COMMIT* aCommit = nullptr );
+    ZONE_FILLER( BOARD* aBoard, COMMIT* aCommit );
     ~ZONE_FILLER();
 
     void InstallNewProgressReporter( wxWindow* aParent, const wxString& aTitle, int aNumPhases );
```
from this commit:
```
commit 463100d67f280c9bd9d32bd01938acf91f93ebb3
Author: Jeff Young <jeff@rokeby.ie>
Date:   Wed Aug 12 19:42:40 2020 +0100

    Remove a long-standing hack to keep divots out of adjacent zones.
    
    The new algorithm unions any adjacent zones before doing the
    chamfer/fillet and then subtracts the other zones back out afterwards.
    
    Fixes https://gitlab.com/kicad/code/kicad/issues/3812
```

NilujePerchut, 2020-10-19:
Hi,
According to this post https://forum.kicad.info/t/yet-another-python-teardrop-script-adds-and-deletes-teardrops-to-a-pcb-v0-3-3/3388/97?u=niluje , the new python API won't be stable before V6-RC1.
Moreover, the old API will be very unstable during this period.
So I will wait for the RC1 before starting to work on this.

Sorry for this bad news.

eeyrjmr, 2021-11-15:
Looping back to this, as this is a fantastic plugin, the API won't be finalised for v6 so what is present now is more than likely what it is on release
https://forum.kicad.info/t/python-api-status-in-v6/31929

NilujePerchut, 2021-11-20:
All 5.99/6_RC1 porting issues are now using the same #35 thread.

NilujePerchut, 2021-11-22:
Hi all,
Please try the V5.99_V6RC branch based on beanjs work.
This was successfully (but quickly) tested on a board of mine (kicad-git/archlinux (V6RC)).
Let me know if you have any comment/bugs

closed 2022-04-16 by NilujePerchut