NilujePerchut/kicad_scripts #31: About teardrop plug-in update to kicad_ V5.99

xiaotaomcu, 2021-06-25:
My friend, your plug-in is easy to use, but it doesn't support v5.99. Do you have time to update the plug-in? Thank you very much

NilujePerchut, 2021-06-29:
This will ported to 5.99 as soon as the Kicad python API is stabilized.

xiaotaomcu, 2021-06-30:
okay, thank you

easyw, 2021-09-21:
it seems it has been patched for k v5.99
https://github.com/beanjs-kicad/beantools/tree/master/beantools/teardrops
it would be nice to have it merged or merged in a dev branch

NilujePerchut, 2021-11-20:
All 5.99/6_RC1 porting issues are now using the same #35 thread.

NilujePerchut, 2021-11-22:
Hi all,
Please try the V5.99_V6RC branch based on beanjs work.
This was successfully (but quickly) tested on a board of mine (kicad-git/archlinux (V6RC)).
Let me know if you have any comment/bugs

closed 2022-04-16 by NilujePerchut