NilujePerchut/kicad_scripts #32: Teardrop on track

Groj7, 2021-08-17:
Hello,

Is it also possible to use teardrops on tracks? 
For example in this situation:
[image]
It would be nice if the tracks are more fluid.
Thank you!

NilujePerchut, 2021-08-29:
Hello,

Sorry for the answer delay.
This is indeed something that could be great (handling the T-Bone track structure would be too!).
I will check what is possible.
But, to be honest, I do not think I will have time to implement it in the near future.

Cédric

Groj7, 2021-08-30:
Hello,

Thank you for your response and no problem! Love the plug-in so far and it all works well. It's funny how KiCAD is getting near Altium level with all these plug-ins. Looking forwards to the future addition. 

Jorg

NilujePerchut, 2022-04-16:
If you need this feature, I suggest you to check at the native Kicad teardrop implementation (V6.99 only).
I might help you.