NilujePerchut/kicad_scripts #33: [bug] plugin keeps KiCad running in the background

T0jan, 2021-09-01:
I recently found this script and its really useful and works quite well but I found one issue: 

If I execute the script as action plugin in KiCad and close KiCad afterwards it still keeps running as a background process and never closes on its own. 

It does not use any resources in this time and you only notice it if you want to start KiCad again and get the warning "KiCad is already running". 

It makes no difference if you actually do something in the plugin or just start it and close it right away, KiCad will continue to run in the background.

my Kicad version:

Application: KiCad
Version: 5.1.10-88a1d61d58~88~ubuntu18.04.1, release build
Libraries:
    wxWidgets 3.0.4
    libcurl/7.58.0 OpenSSL/1.1.1 zlib/1.2.11 libidn2/2.0.4 libpsl/0.19.1 (+libidn2/2.0.4) nghttp2/1.30.0 librtmp/2.3
Platform: Linux 5.4.0-81-generic x86_64, 64 bit, Little endian, wxGTK
Build Info:
    wxWidgets: 3.0.4 (wchar_t,wx containers,compatible with 2.8) GTK+ 3.22
    Boost: 1.65.1
    OpenCASCADE Community Edition: 6.9.1
    Curl: 7.58.0
    Compiler: GCC 7.5.0 with C++ ABI 1011

Build settings:
    USE_WX_GRAPHICS_CONTEXT=OFF
    USE_WX_OVERLAY=ON
    KICAD_SCRIPTING=ON
    KICAD_SCRIPTING_MODULES=ON
    KICAD_SCRIPTING_PYTHON3=ON
    KICAD_SCRIPTING_WXPYTHON=ON
    KICAD_SCRIPTING_WXPYTHON_PHOENIX=ON
    KICAD_SCRIPTING_ACTION_MENU=ON
    BUILD_GITHUB_PLUGIN=ON
    KICAD_USE_OCE=ON
    KICAD_USE_OCC=OFF
    KICAD_SPICE=ON

NilujePerchut, 2021-10-03:
@easyw : I saw your comment right after pushing a similar solution. Nevertheless, if you think yours is better, just tell me and I will revert mine to implement yours

T0jan, 2021-10-19:
Thanks for solving this bug. do you want this to stay open with the last question or can it be closed?

easyw, 2021-11-01:
@NilujePerchut 
thanks for this useful plugin 😸