NilujePerchut/kicad_scripts #35: Not working on kicad6, development version of kicad

sahil-kalka, 2021-11-18:
I am trying to get it working on kicad nightly development version but failed, can you please make another branch of this plugin for kicad 6

pdp7, 2021-11-18:
FYI - this was posted recently by KiCad developer Jon Evans:
[Calling all Python devs - KiCad 6 and plugin repository information](https://lists.launchpad.net/kicad-developers/msg45427.html)
> We need your help to test out your plugins and make any changes needed for the 6.0 API, so that we can identify and fix any remaining regressions from 5.1 before the December 15 deadline. If you have any trouble getting yourscripts to work in 6.0 RC1, please open an issue on GitLab and we will assist.  It is important to us that existing plugins are able to continue to work.

NilujePerchut, 2021-11-20:
Hi all.
Since the official dev team choice is not to implements the new Python API for RC1, I will create a dedicated 6_RC1 branch and do the necessary changes.

I will also choose this thread to merge all V6_RC1/5.99 porting open issues (#26 and #31).

NilujePerchut, 2021-11-22:
Hi all,
Please try the V5.99_V6RC branch based on beanjs work.
This was successfully (but quickly) tested on a board of mine (kicad-git/archlinux (V6RC)).
Let me know if you have any comment/bugs

sahil-kalka, 2021-11-22:
Hi,

It seems same as earlier version, It is also showing errors while initializing. For me its unable to initialize and again I need to do some modification to make it visible to kicad.

sahil-kalka, 2021-11-22:
Seems it is working, I saw a bug there. for curved traces it not working fine. See the image, its self explanatory. 

[image]

NilujePerchut, 2021-11-22:
Thanks for the test.
Since curved tracks are new to V5.99/V6RC I did not had time test them against this plugin.
I will try to fix this as soon as possible.
Is it possible for you to sent me the faulty design in order for me to confirm that my fix is OK.

sahil-kalka, 2021-11-23:
Hi, please find PCB file as attached
[AC_Controller_21.zip](https://github.com/NilujePerchut/kicad_scripts/files/7585993/AC_Controller_21.zip)
.

pierrotm777, 2021-12-29:
Hello,
*I need help for install your plugin.
It worked fine with kicad 5.1 but i don't understand how to install it with kicad 6.0.
Thanks for your help
Pierre

ajinkya-open, 2022-01-05:
I am everyday user of Kicad since 2 year, i have instlled new viersion but i am also stuck how to install this plugin following is the logs

`  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1014, in _gcd_import
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 848, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/usr/share/kicad/scripting/plugins/teardrops/__init__.py", line 1, in <module>
    from .teardrop_plugin import TeardropPlugin
  File "/usr/share/kicad/scripting/plugins/teardrops/teardrop_plugin.py", line 14, in <module>
    from .teardrop_dialog import InitTeardropDialog
  File "/usr/share/kicad/scripting/plugins/teardrops/teardrop_dialog.py", line 16, in <module>
    from .td import SetTeardrops, RmTeardrops, __version__
  File "/usr/share/kicad/scripting/plugins/teardrops/td.py", line 12, in <module>
    from pcbnew import VIA, ToMM, TRACK, FromMM, wxPoint, GetBoard, ZONE_CONTAINER
ImportError: cannot import name 'VIA' from 'pcbnew' (/usr/lib/python3/dist-packages/pcbnew.py)`

sahil-kalka, 2022-01-05:
Hey, please use 6.0 branch to see things happening.

WhiteChairFromIkea, 2022-02-10:
> I have it added to my plugin folder, but it does not show up on the menu at all.. any suggestions ?

https://forum.kicad.info/t/is-there-a-teardrop-plugin-going-for-v6-0/33141/5?u=poco

indymx, 2022-02-10:
> > I have it added to my plugin folder, but it does not show up on the menu at all.. any suggestions ?
> 
> https://forum.kicad.info/t/is-there-a-teardrop-plugin-going-for-v6-0/33141/5?u=poco

Yep, that's what I did.. Does not show on the menu..

indymx, 2022-02-10:
`terry@serge:~/.local/share/kicad/6.0/scripting/plugins$ ls -la
total 0
drwxrwxr-x 1 terry terry 130 Feb 10 08:11 .
drwxrwxr-x 1 terry terry  14 Dec 25 16:57 ..
drwxrwxr-x 1 terry terry 240 Feb  9 05:23 InteractiveHtmlBom
drwxrwxr-x 1 terry terry 170 Feb  9 05:53 kicad-action-scripts
drwxrwxr-x 1 terry terry 378 Feb  9 05:53 kicad-round-tracks
drwxrwxr-x 1 terry terry 230 Feb  9 05:52 teardrops
terry@serge:~/.local/share/kicad/6.0/scripting/plugins$ cd teardrops/
terry@serge:~/.local/share/kicad/6.0/scripting/plugins/teardrops$ ls -la
total 132
drwxrwxr-x 1 terry terry   230 Feb  9 05:52 .
drwxrwxr-x 1 terry terry   130 Feb 10 08:11 ..
-rw-rw-r-- 1 terry terry    72 Mar 10  2021 __init__.py
drwxrwxr-x 1 terry terry   508 Feb  9 06:01 __pycache__
drwxrwxr-x 1 terry terry   100 Mar 10  2021 rcs
-rw-rw-r-- 1 terry terry  1950 Mar 10  2021 README.md
-rwxrwxr-x 1 terry terry 13208 Feb  9 05:52 td.py
-rw-rw-r-- 1 terry terry  2819 Feb  9 05:52 teardrop_dialog.py
-rw-rw-r-- 1 terry terry 78180 Feb  9 05:52 teardrop.fbp
-rw-rw-r-- 1 terry terry  5300 Feb  9 05:52 teardrop_gui.py
-rw-rw-r-- 1 terry terry   805 Mar 10  2021 teardrop_plugin.py
-rw-rw-r-- 1 terry terry  9409 Mar 10  2021 teardrops.png
terry@serge:~/.local/share/kicad/6.0/scripting/plugins/teardrops$`

climent, 2022-02-10:
Move the zip file one level higher in the tree hierarchy and unpack it there. I followed the instructions and that fixed it.

indymx, 2022-02-10:
If I were to do that, it'd be outside of the plugins folder. It is currently in:

/home/terry/.local/share/kicad/6.0/scripting/plugins/teardrops

I have InteractiveBOM, Round Tracks CIrcularZone and ViaStitching in the plugins folder, all work just fine. 

terry@serge:~/.local/share/kicad/6.0/scripting/plugins$ ls
InteractiveHtmlBom  kicad-action-scripts  kicad-round-tracks  teardrops

> Move the zip file one level higher in the tree hierarchy and unpack it there. I followed the instructions and that fixed it.

climent, 2022-02-10:
What do you have inside the terry@serge:~/.local/share/kicad/6.0/scripting/plugins/teardrops directory?

WhiteChairFromIkea, 2022-02-10:
> The contents of the zip
> […](#)

Screenshot, please

WhiteChairFromIkea, 2022-02-10:
> The contents of the zip
> […](#)

Screenshot, please..

indymx, 2022-02-10:
Came home to a DOA Linux boot partition.. Here's the screeny from within my
windows boot.. still the same drive though..

[image: image.png]

climent, 2022-02-11:
Do you see this?
climent-macmini:~ climent$ ls Documents/KiCad/6.0/scripting/plugins/teardrops/
LICENSE		README.md	__init__.py	__pycache__	teardrops

climent, 2022-02-11:
> Came home to a DOA Linux boot partition.. Here's the screeny from within my windows boot.. still the same drive though.. [image]

That's not the right content. On the plugins directory, remote the teardops directory and subdirectories, and then unzip this file https://github.com/NilujePerchut/kicad_scripts/archive/refs/heads/V5.99_V6RC.zip

WhiteChairFromIkea, 2022-02-11:
And look at my previous post, there are screnshots with correct contents for a reference.

closed 2022-04-16 by NilujePerchut