NilujePerchut/kicad_scripts #40: PCBNEW seg fault after Add or Remove teardrop on Kicad6

NilujePerchut, 2022-01-09:
Using V5.99 and Kicad 6.0.0, removing or adding teardrops makes pcbnew segfault when closing.

Steps to reproduce:
1) Open a design
2) Add or remove teardrops
3) Close pcbnew
==> seg fault

Remarks:
Also happens with #39 and #37 merged.

mitxela, 2022-02-12:
I've been seeing a similar problem with the round tracks plugin. It only happens when items are removed from the board, and I think it only happens when you close the window with unsaved changes.

I don't know if both plugins have exactly the same issue, but I've filed an issue here: https://gitlab.com/kicad/code/kicad/-/issues/10827

I've been using my branch of the repo (#41) without problems, the segfault only happens for me if I remove teardrops and close the window without saving.

NilujePerchut, 2022-02-19:
Thanks for help. I also hope that this a common issue.
For the record, I also encountered this bug after adding teardrops. But this was right at the kicad 6.0 release. It might have changed since.

closed 2022-04-16 by NilujePerchut