NilujePerchut/kicad_scripts #43: Teardrops not added to all pads

perdrix52, 2022-06-06:
Just installed this PI (0.6.0).   Most pads did have teardrops added, but some didn't:

[image]

BTW can you explain what  "Do not  insert teardrops inside same net zones" means?

Thanks
D.

NilujePerchut, 2022-06-06:
Teardrops on SMD pad in not well supported. You have to explicitly check the `Include SMD pads` if you want to try it.

The `Do not insert teardrops inside same net zones` prevents you to add teardrops inside a kicad zone. This is usually the good behavior to avoid holes in zone. See issue #18 for more details on that.

perdrix52, 2022-06-06:
I did tick Include SMD pads

NilujePerchut, 2022-06-06:
As I said above, teardrops on SMD pad are not working properly.
If you really need this feature, I suggest you to take look at the new kicad builtin teardrop (needs to compile the current nightly with a special flag).

WhiteChairFromIkea, 2022-06-06:
> As I said above, teardrops on SMD pad are not working properly. If you really need this feature, I suggest you to take look at the new kicad builtin teardrop (needs to compile the current nightly with a special flag).

It is easier than that, no need to compile anything: please download nightly (v6.99) and install as usual (will be side by side with existing v6.0). Then set allowteardrops flag insidde kicad_advanced file (search for that in the kicad forum). Teardrops are more advanced in native kicad, and I had no issues, ecen track neck down is implemented... But, dragons ahead, f8les saved with nightly (v6.99) are not backwards compatible with v6.0, so do backups.

NilujePerchut, 2022-06-06:
Thanks a lot ! I will give it a try.